analogdevicesinc/hdl #138: spi engine address decoding mismatch

mxdsgnl, 2018-07-31:
Hello,

https://wiki.analog.com/resources/fpga/peripherals/spi_engine/axi shows that the SDI "Peek Register" address is at <Base_Address>+ 0xEC in PS memory space.  

axi_spi_engine.v line 309 shows that the peek register is decoded at 0x3C which maps to offset 0xF0 from the PS.

Thanks,
Nick

lclausen-adi, 2018-07-31:
Hi,

Thanks for letting us know. I think the intention for that register was to be add address 0x3b. But at this point the best approach is probably just to update the documentation to match the implementation.

mxdsgnl, 2018-07-31:
You're welcome.  Thanks for your response.

Nick

closed 2018-08-01 by lclausen-adi

rgetz, 2018-08-01:
I thought @rkutty had some automated scripts that generated the wiki docs. Isn't @acostina using these to update the wiki?

acostina, 2018-08-02:
We're using the scripts to generate the common register map (ADC/DAC) + DMA register map, based on some files which contain the description of the registers. We don't extract the registers from verilog, the files are modified manually. We'll integrate the JESD204 and the SPI Engine in the next few months, but still, the process will be partially manual, and when we change the code we need to make sure we also change those files.

acostina, 2018-08-02:
Currently, the scripts generate a markup compatible with the wiki and, in the DMAC case, compatible with the testing environment. The next generation scripts will create markup compatible with reStructuredText, which will be part of the hdl repository, docs folder.  

The wiki will point to the hdl docs, but these changes will be available in a few months I think,  after we have a pilot project tested and see how it looks and feels.

rgetz, 2018-08-02:
Sounds great, as long as there is a plan.

Thanks