analogdevicesinc/hdl #193: vpll usage and risk

vlsireddy, 2018-10-11:
Hi,
   in the zcu102 block diagram, vpll is being used to drive TOP_SW_MAIN switch.

these are PS switches within the MPSOC,

https://www.xilinx.com/support/answers/69685.html
there is this AR from xilinx, which says, there might be issues 
1. in linux 2017.1
2. when using display port,

would you like to change the TOP_SWMAIN source from vpll to iopll??

Csomi, 2018-10-12:
Hi,

We do not support 2017.1, and if I understand correctly this issue is fixed after 2017.2. 
The latest version which we're supporting is 2017.4.1.

Did you validate this issue on 2017.4? 

I'm also adding @mhennerich to the discussion, being this issue more linux related than hdl.

Csomi, 2018-12-13:
Will close this issue, feel free to re-open it if you have anything more to add.

closed 2018-12-13 by Csomi