analogdevicesinc/hdl #196: axi_dmac bugs

xiaoweihua, 2018-10-12:
1. the maximum transfer length is 262140, if exceeding this number, the transfer can not terminated.
2. the destination axi bus shows data overflow when transfer
[image]
[image]

lclausen-adi, 2018-10-15:
Hi,
Can you tell us a few more details about your setup. Which version of the HDL are you using, which project and what is the configuration of the DMAC.

The overflow condition in the ILA seems to occur when the internal tracking memory of the ILAS is full. In this case it is recommended to reconfigure the ILA with an increase number of outstanding IDs. See https://www.xilinx.com/support/documentation/sw_manuals/xilinx2018_2/ug908-vivado-programming-debugging.pdf#page=209

Thanks.

lclausen-adi, 2018-10-15:
Support for this will continue at https://ez.analog.com/fpga/f/q-a/102388/the-maximum-transfer-length-of-axi_dmac

closed 2018-10-15 by lclausen-adi