analogdevicesinc/hdl #217: 2018_r2 generated device tree for AD9361 is not working

aymanhendawy, 2018-12-23:
Hi,

I'm working on 2018_r2, "/projects/fmcomms2/zcu102" to generate system.hdf, I used this file once with petalinux and also with vivado 2018.2 to generate a device tree .dts and .dtb, in both trials I used linux_adi kernel source, both ways (petalinux and vivado) failed to define the AD9361 to the kernel through the generated dts dtb files, I got to this conclusion becasue when I ran iio_info | grep iio:device, I can't found the ad9361 device listed, but when I put the zynqmp-zcu102-rev10-ad9361-fmcomms2-3.dtb file generated from the makefile of linux_adi repo it works fine, and got my device listed.

Csomi, 2018-12-23:
Please post your issue on Engineer Zone (https://ez.analog.com/linux-device-drivers/linux-software-drivers). And expect some delays due the holidays. Thanks for understanding.

jiajiahao0619, 2019-01-01:
Hi,
I use the branch of 2016_r2, I can generate the bit and launch the SDK. I follow the steps of github. When I download the bit by SDK. The hdmi monitor display the right picture just 1 or2 seconds. Then it turn back or flash. I do not know the reasion.
The uart display:
MSEN changed to HI
HPD changed to LOW
MSEN changed to HI
I2C timeout.
            I2C timeout.
                        I2C timeout.
                                    I2C timeout.
                                                HPD changed to LOW
MSEN changed to HI
HPD changed to LOW
MSEN changed to HI
HPD changed to LOW
MSEN changed to HI
I2C timeout.
            I2C timeout.
                        I2C timeout.
                                    I2C timeout.
                                                A new EDID segment was read.
DVI device.
------------------------- EDID BLOCK 0 -------------------------
Checksum failed on block 0.

please help me.

closed 2019-03-11 by Csomi