analogdevicesinc/hdl #326: Add default peripherals to Pluto PS PCB test points

nunojpg, 2019-08-04:
The Pluto PCB exposes 7 PS pins which are unconfigured by default.

For user applications to use them a reconfigured HDL is required.

Could this HDL expose them by default as some mix of peripherals? Eg. 7 x GPIO or 5x GPIO + 1 x I2C.

If this does not have any bad side effects then it would make basic user mods easier.

acostina, 2019-08-20:
Hello,

The PS GPIOs (MIO GPIO) are enabled by default in our HDL for all the 7 exposed pins

nunojpg, 2019-08-20:
Does this mean only the device tree need to be modified to load the linux drivers?

acostina, 2019-08-21:
For GPIOs you should be able to just export the desired GPIO and drive it directly from linux. On a test pluto I have, I am exporting 906 for PS_GPIO_0 using:

cd /sys/class/gpio
echo 906>export

closed 2019-08-28 by acostina