analogdevicesinc/hdl #419: [BUG] Compilation fails in Pluto project.

MichalMisiaszek, 2020-01-17:
**Describe the bug**
Can't compile with 2018.3 on Ubuntu 16.04

**To Reproduce**
Steps to reproduce the behavior:
1. /hdl
2. make all
3. wait
4. See error

**Expected behavior**
The successful compilation of code.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - ZC102
 - Ubuntu 16.04
 - Vivado 2018.3
 - Head of master
** Error:
Building fmcadc5_vc707 project [/home/michal/user_ip_ex/hdl/projects/fmcadc5/vc707/fmcadc5_vc707_vivado.log] ... FAILED
For details see /home/michal/user_ip_ex/hdl/projects/fmcadc5/vc707/fmcadc5_vc707_vivado.log

Log:

CRITICAL WARNING: vivado version mismatch; expected 2019.1, got 2018.3.
WARNING: [Device 21-436] No parts matched 'xc7vx485tffg1761-2'
ERROR: [Coretcl 2-106] Specified part could not be found.
INFO: [Common 17-206] Exiting Vivado at Fri Jan 17 14:44:16 2020...

tfcollins, 2020-01-18:
As the error log says you are using the wrong version of vivado. Release 2019_R1 supports Vivado 2018.3. Try building with that branch (hdl_2019_r1).

-Travis

MichalMisiaszek, 2020-01-18:
Thank you I just found it. I need to pull the release you indicated.

MichalMisiaszek, 2020-01-18:
I pulled the correct version but still fails:
# adi_project fmcadc5_vc707
WARNING: [Device 21-436] No parts matched 'xc7vx485tffg1761-2'
ERROR: [Coretcl 2-106] Specified part could not be found.
INFO: [Common 17-206] Exiting Vivado at Fri Jan 17 16:18:35 2020...

Csomi, 2020-01-18:
Most likely, you did not select 7 Series device support when you were installing Vivado. If you want to build all the projects, make sure that you install all the device types.

MichalMisiaszek, 2020-01-18:
Yes, you absolutely right. In order to save space I installed only specific devices. I am using evaluation board ZC102. Can I build project for selected devices only?

Csomi, 2020-01-18:
Just run make in the project's directory. E.g. cd projects/fmcomms2/zc102; make

MichalMisiaszek, 2020-01-18:
Thanks for clarification, this is not a bug in fact. It would be nice to have some "configure" script to detect installed devices and report back to user or not compile missing ones.

closed 2020-01-18 by MichalMisiaszek