analogdevicesinc/hdl #427: Error while building  fmcjesdadc1  project

diegovelayarce, 2020-02-04:
Hello,
I am trying to build a new project from the repository and I am currently working with the master branch and vivado 2018.3 when I try to run the make file I get the followning error:

Building fmcjesdadc1_zc706 project [/home/soporte/adi/hdl/projects/fmcjesdadc1/zc706/fmcjesdadc1_zc706_vivado.log] ... FAILED
For details see /home/soporte/adi/hdl/projects/fmcjesdadc1/zc706/fmcjesdadc1_zc706_vivado.log

../../scripts/project-xilinx.mk:58: recipe for target 'fmcjesdadc1_zc706.sdk/system_top.hdf' failed
make: *** [fmcjesdadc1_zc706.sdk/system_top.hdf] Error 1
****

I have tried changing the branch to the hdl_2019_r1 but I still get the same error.
Thanks.

Csomi, 2020-02-04:
You have to use hdl_2019r1 branch with Vivado 2018.3. Did you try to check the log file? What does it say?

diegovelayarce, 2020-02-04:
I changed the branch and checked the log and it shows the following:
# adi_project fmcjesdadc1_zc706
WARNING: [Device 21-436] No parts matched 'xc7z045ffg900-2'
ERROR: [Coretcl 2-106] Specified part could not be found.
INFO: [Common 17-206] Exiting Vivado at Tue Feb  4 14:52:00 2020...

acostina, 2020-02-05:
You don't seem to have installed the FPGA support for the FPGA on the ZC706 with your Vivado. ZC706 requires an license, it doesn't work with webpack.

diegovelayarce, 2020-02-05:
Thanks for the replies it was a license issue indeed.

closed 2020-02-05 by diegovelayarce