analogdevicesinc/hdl #489: where can i download the Schematic  related to the program

lxdfrank, 2020-05-21:
Thank you for your code demo about chiped of adi and it improved the efficiency. But I can find the related schematic of verilog code. and can you provide the schematic of connection for me?

Best regards

Csomi, 2020-05-21:
We are not using schematic to design our projects. In both Vivado and Quartus we are using tcl scripts like [this](https://github.com/analogdevicesinc/hdl/blob/master/projects/common/zc706/zc706_system_bd.tcl) or [this](https://github.com/analogdevicesinc/hdl/blob/master/projects/fmcomms2/common/fmcomms2_bd.tcl) to define the configuration of various IP instances and to define the systems' block design (interconnection between IPs). 

You can find more info on [HDL User Guide](https://wiki.analog.com/resources/fpga/docs/hdl).

Also, for further questions please open up a thread at [Engineer Zone](https://ez.analog.com/fpga). Github issues are for bugs only.

lxdfrank, 2020-05-21:
Okay， thank you a lot for your answer.

closed 2020-05-21 by Csomi