analogdevicesinc/hdl #529: [BUG]linux正常启动后，拔掉所有micro usb后就没画面了

LiJiaDong0922, 2020-08-13:
linux正常启动后，拔掉所有micro usb后就没画面了

LiJiaDong0922, 2020-08-13:
使用的是zedboard开发板

Csomi, 2020-08-13:
Hi @LiJiaDong0922,
The preferred spoken language in this repository is English. Please use that.
Thank you for your understanding.
Best regards,
-István

LiJiaDong0922, 2020-08-14:
I used the VIvado version of PL file and downloaded HDL-2018_R1
Kernel USES is the kernel of https://github.com/analogdevicesinc/linux.git
U - the use of the boot is https://github.com/Xilinx/u-boot-xlnx v2017.4 version
The file system USES version LINaro20111219
I ran the Linux graphical interface on the Zedbard development board. After unplugging all microusB, there was no image output. When I plugged in MicrousB again, there was image output.

Csomi, 2020-08-15:
For Linux related issue please open up a thread here: https://ez.analog.com/linux-software-drivers/ 

I'm closing this issue now. Thanks.

closed 2020-08-15 by Csomi