analogdevicesinc/hdl #598: SPI connections on zcu102/fmcomms8 don't look right

aholtzma, 2020-10-28:
For fmcomms8, a small bit logic is used to detect a transaction to the 3-wire SPI device and route spi_miso appropriately. However the slave selects going into this logic don't look right: 

https://github.com/analogdevicesinc/hdl/blob/04a694251e62a328ddf4857739881ca29722e7a9/projects/fmcomms8/zcu102/system_top.v#L163-L170

spi_csn is the three spi slave selects concatenated together using xlconcat in the block diagram. This binary to one-hot is not necessary.

aholtzma, 2020-10-28:
I think fmcomm8_spi is ok, it just wants a vector of slave selects.

From this I gather that the fmcomms8/zcu102 hardware combination has not yet been tested. Do you have an idea of when this will happen?

acostina, 2020-10-28:
fmcomms8/zcu102 works as it is, why do you think it doesn't work ?

aholtzma, 2020-10-28:
spi_csn is a vector of chip selects, not a binary encoded chip select. So if slave 0 was selected it would be 3'b110 and  spi_3_to_8_csn would be decoded as 8'b11111111 and there would be no slave selects out to the devices because they are driven from it:

```
 assign spi_csn_adrv9009_c = spi_3_to_8_csn[0];
 assign spi_csn_adrv9009_d = spi_3_to_8_csn[1];
 assign spi_csn_hmc7044 = spi_3_to_8_csn[2];
```

acostina, 2020-10-28:
Linux sets it as a encoded chips select at:
https://github.com/analogdevicesinc/linux/blob/master/arch/arm64/boot/dts/xilinx/zynqmp-zcu102-rev10-adrv9009-fmcomms8.dts#L20

aholtzma, 2020-10-28:
Ok this is the piece I was missing. I was not aware you could even do this.

aholtzma, 2020-10-28:
Added a comment in #599. Thanks for your help.

closed 2020-10-28 by aholtzma