analogdevicesinc/hdl #612: [BUG] make doesn't work for git tag 2019_r2 project /hdl/projects/fmcomms8/zcu102

JanVerlinden, 2020-11-30:
✔ ~/projects/AD-FMCOMMS8-EBZ_ecd880d4_2019_r2/hdl/projects/fmcomms8/zcu102 [2019_r2|✔] 
14:30 $ make
Building util_cdc library [/home/zynqdev/projects/AD-FMCOMMS8-EBZ_ecd880d4_2019_r2/hdl/library/util_cdc/util_cdc_ip.log] ... FAILED
For details see /home/zynqdev/projects/AD-FMCOMMS8-EBZ_ecd880d4_2019_r2/hdl/library/util_cdc/util_cdc_ip.log

../scripts/library.mk:70: recipe for target 'component.xml' failed
make[3]: *** [component.xml] Error 2
../scripts/library.mk:77: recipe for target 'xilinx_dep' failed
make[2]: *** [xilinx_dep] Error 2
../scripts/library.mk:77: recipe for target 'xilinx_dep' failed
make[1]: *** [xilinx_dep] Error 2
../../scripts/project-xilinx.mk:76: recipe for target 'lib' failed
make: *** [lib] Error 2
✘-2 ~/projects/AD-FMCOMMS8-EBZ_ecd880d4_2019_r2/hdl/projects/fmcomms8/zcu102 [2019_r2|✔] 
14:30 $ git checkout 19b7986486a56bd3282187951b0836062b693a19
Previous HEAD position was ecd880d4 adrv9009zu11eg:fmcomms8: Fix SPI timing constraint
HEAD is now at 19b79864 fmcomms8: Fix SPI timing
✔ ~/projects/AD-FMCOMMS8-EBZ_ecd880d4_2019_r2/hdl/projects/fmcomms8/zcu102 [:19b79864|✔] 
14:36 $ make
Building ad_ip_jesd204_tpl_adc library [/home/zynqdev/projects/AD-FMCOMMS8-EBZ_ecd880d4_2019_r2/hdl/library/jesd204/ad_ip_jesd204_tpl_adc/ad_ip_jesd204_tpl_adc_ip.log] ... OK
Building ad_ip_jesd204_tpl_dac library [/home/zynqdev/projects/AD-FMCOMMS8-EBZ_ecd880d4_2019_r2/hdl/library/jesd204/ad_ip_jesd204_tpl_dac/ad_ip_jesd204_tpl_dac_ip.log] ...

tfcollins, 2020-12-01:
Can you provide the log file?
-Travis

JanVerlinden, 2020-12-01:
I think you can close this topic again, I wasn't aware that the vivado version was upgraded to 2019.1.
The 2019_r2 is not in the list(https://wiki.analog.com/resources/fpga/docs/releases). So I assumed that vivado stayed on 2018.3.

ERROR: vivado version mismatch; expected 2019.1, got 2018.3.
This ERROR message can be down-graded to CRITICAL WARNING by setting ADI_IGNORE_VERSION_CHECK environment variable to 1. Be aware that ADI will not support you, if you are using a different tool version.
INFO: [Common 17-206] Exiting Vivado at Tue Dec  1 08:40:00 2020...

closed 2020-12-01 by tfcollins