analogdevicesinc/hdl #639: adrv9361z7035 axi_ad9361 clock output incorrect frequency

matthewspydell, 2021-01-24:
Commit [4b13274c55bf1f899867ce402a581390ddd28235](https://github.com/analogdevicesinc/hdl/commit/4b13274c55bf1f899867ce402a581390ddd28235#diff-d0f0d311ad4bf2826e7a825c19e54a20e80a1f1a4e3ad01d1ead83b1c17970c6) removed a `create_clock` command on the `axi_9361` module's clock. Instead, just the incoming clock diff pair has a `create_clock` command now.

This is understandable as the incoming clock pair goes into a clock buffer with the output coming out of the `axi_ad9361` module, and it's this clock that used to have the other `create_clock` command. However, I am observing a behavior where Vivado is listing the clock coming out of the `axi_ad9361` module as being a 100 MHz clock even though the diff pair driving the clock buffer is set to be a 250 MHz clock. I can edit the frequency Vivado thinks the clock is from 100 MHz to 250 MHz, but I want to know if this is a known issue or if I am missing something here.

matthewspydell, 2021-01-26:
The fix is to add this to the [`axi_ad9361_ip.tcl`](https://github.com/analogdevicesinc/hdl/blob/master/library/axi_ad9361/axi_ad9361_ip.tcl#L98) script starting at line 98:
```tcl
ipx::add_bus_parameter FREQ_HZ [ipx::get_bus_interfaces l_clk -of_objects [ipx::current_core]]
set_property value 250000000 [ipx::get_bus_parameters FREQ_HZ -of_objects [ipx::get_bus_interfaces l_clk -of_objects [ipx::current_core]]]
```

sarpadi, 2021-01-27:
Hello,
The [commit](https://github.com/analogdevicesinc/hdl/commit/4b13274c55bf1f899867ce402a581390ddd28235#diff-d0f0d311ad4bf2826e7a825c19e54a20e80a1f1a4e3ad01d1ead83b1c17970c6) you are referring to is needed for timing. The fact that vivado identifies l_clk as 100MHz only has to do with the block design and does not affect timing (you can see this in post-implementation), it is simply a "cosmetic" issue at this point and it most certainly can be found in other projects as well.
The fix you are suggesting is not the best idea since it will apply to all the instances of this particular IP core and it will also only fix part of the problem. The correct approach would be to package the IP so that the input clock frequency (rx_clk_in_p) would be automatically identified and propagated to the l_clk output.
~Sergiu

matthewspydell, 2021-01-27:
I was thinking it was just "cosmetic" as you say. Since I am modifying the design a fair amount for my own purposes, I don't need to worry about any other instances of that particular IP, but I understand your point. Thanks for the info.

closed 2021-01-27 by matthewspydell