analogdevicesinc/hdl #685: Which value should I use in "axi_ddr_cntrl"

liubenyuan, 2021-04-20:
Hi, when I recreate the `project/ad9081_fmca_ebz/vcu118` and error occured,

```tcl
### ad_ip_parameter axi_ddr_cntrl CONFIG.C0_CLOCK_BOARD_INTERFACE default_250mhz_clk2
### ad_ip_parameter axi_ddr_cntrl CONFIG.C0_DDR4_BOARD_INTERFACE ddr4_sdram_c2
ERROR: [IP_Flow 19-3461] Value 'ddr4_sdram_c2' is out of the range for parameter 'C0 DDR4 BOARD INTERFACE(C0_DDR4_BOARD_INTERFACE)' for BD Cell 'axi_ddr_cntrl' . Valid values are - Custom, ddr4_sdram_c1_062, ddr4_sdram_c1_083, ddr4_sdram_c2_062, ddr4_sdram_c2_083
INFO: [IP_Flow 19-3438] Customization errors found on 'axi_ddr_cntrl'. Restoring to previous valid configuration.
ERROR: [BD 41-245] set_property error - Value 'ddr4_sdram_c2' is out of the range for parameter 'C0 DDR4 BOARD INTERFACE(C0_DDR4_BOARD_INTERFACE)' for BD Cell 'axi_ddr_cntrl' . Valid values are - Custom, ddr4_sdram_c1_062, ddr4_sdram_c1_083, ddr4_sdram_c2_062, ddr4_sdram_c2_083
Customization errors found on 'axi_ddr_cntrl'. Restoring to previous valid configuration.

ERROR: [Common 17-39] 'set_property' failed due to earlier errors.
```

It seems that I should set a value for `axi_ddr_cntrl`, which value should I use? `ddr4_sdram_c2_062` or `ddr4_sdram_c2_083`

acostina, 2021-04-20:
Hello, 
This typically happens when a different version of Vivado is used than the one we support. Are you using Vivado 2020.1 (https://github.com/analogdevicesinc/hdl/blob/master/projects/scripts/adi_project_xilinx.tcl#L4) ?

liubenyuan, 2021-04-20:
Yes, I am using vivado 2020.1.1, I modify the scripts in `projects/scripts/adi_project_xilinx.tcl` and `library/scripts/adi_ip_xilinx.tcl` and using the make scripts, and the project compiles OK. Thanks.

closed 2021-04-20 by liubenyuan