analogdevicesinc/hdl #795: [BUG] vivado is teminated. When adding axi_adxcvr IP in the kria k26c board file environment.

gudokim, 2021-11-05:
**Describe the bug**
I made custom kria board and it has FMC to interface the DAQ2 board.
And working on migrating the daq2_zcu102 to the kria k26c som.

I followed a guideline to build the libraries in tcl console on vivado.
https://wiki.analog.com/resources/fpga/docs/build#xilinx_auto_tcl_build

After build all of libraries for DAQ2, added them to the IP repositories.
When adding axi_adxcvr IP in the kria k26c board file environment, vivado is teminated.

For test, I created the new project with the kria k26c board file and added HDL IP libraries folder to vivado.
If tried to add the axi_adxcvr IP even in the empty block design, vivado is teminated.
Other IPs seemed no problem.

** Screen shot **
[image]

**To Reproduce**
Steps to reproduce the behavior : On the vivado tcl console
1. create_project project_7 C:/xilinx/hdl/projects/daq2/zcu102/project_7 -part xck26-sfvc784-2LV-c
2. set_property  ip_repo_paths  C:/xilinx/hdl/library [current_project]
3. create_bd_design "design_1"
4. create_bd_cell -type ip -vlnv analog.com:user:axi_adxcvr:1.0 axi_adxcvr_0
5. vivado is teminated

**Expected behavior**
A clear and concise description of what you expected to happen.

**Desktop (please complete the following information):**
 - Project name and used carrier board [DAQ2 K26C]
 - Used Software: [win10]
 - Tool version [Vivado 2021.1]
 - HDL Release version [master]

**Additional context**
Add any other context about the problem here.

ronagyl, 2021-11-05:
Please could you test the following fix? 
 https://github.com/analogdevicesinc/hdl/pull/796

gudokim, 2021-11-05:
It works like a charm.
Thanks a lot.

closed 2021-11-05 by gudokim