analogdevicesinc/hdl #838: hold timing problem

drorar, 2022-01-06:
**Describe the bug**
Hello

In the AD9361 (Xilinx -- Vivado 2021.2) I get hold time issue between up_d_count and d_count in the up_clock_mon file
The d_count is generate in a different clock then the up_d_count

**To Reproduce**

[image]

**Desktop (please complete the following information):**
 - Project name and used carrier board [e.g. FMCOMMS2 ZC706]
 - Used Software: [e.g. Linux or No-OS]
 - Tool version [e.g. Vivado 2018.3, Quartus PRO 18.1]
 - HDL Release version [e.g. hdl_2019_r1]
 - Software Release version [e.g. 2019_R1]

**Additional context**
Add any other context about the problem here.

jamesoncollins, 2023-11-13:
I've also seen issues with this module.  The clock monitor appears to have synchronization registers but doesn't have constraints, or perhaps ASYNC_REG attributes, to tell the tool its an async clock path.  For some reason I don't get a timing error, just a critical warning.

angelgarg, 2024-10-17:
pls assign me task under hactoberfest