analogdevicesinc/hdl #899: [BUG] Undefined fpga technology for "xck26-sfvc784-2LV-c"

gitmodimo, 2022-03-29:
**Describe the bug**
While instantiating ADRV9001 ip in non-project mode error occours:
Undefined fpga technology for "xck26-sfvc784-2LV-c"

We tracked down the issue to library\scripts\adi_xilinx_device_info_enc.tcl file.
It seem FPGA_TECHNOLOGY regexp does not handle new Xilinx Kria SOM

```
      FPGA_TECHNOLOGY {
          switch  -regexp -- $part {
             ^xc7    {set series_name 7series}
             ^xczu   {set series_name ultrascale+}
             ^xc.u.p {set series_name ultrascale+}
             ^xc.u   {set series_name ultrascale }
             default {
                 puts "Undefined fpga technology for \"$part\"!"
                 exit -1
             }
          }
          return "$series_name"
      }
```

Adding additional case fixes this problem:
`^xck26 {set series_name ultrascale+}`

ronagyl, 2022-03-29:
This should be part of the master branch already. 
https://github.com/analogdevicesinc/hdl/blob/master/library/scripts/adi_xilinx_device_info_enc.tcl#L148
Please can you double check ? 

Thanks.

gitmodimo, 2022-03-29:
It looks like you are correct. Exactly the same fix is applied. We are currently using hdl_2019_r2 branch like  https://github.com/analogdevicesinc/hdl/releases states. Is there any newer stable branch? Or should we use master?

ronagyl, 2022-03-29:
You can continue using the 2019_r2  with the mentioned changes.  Master may be unstable sometimes. 
I will close this bug with this resolution.

closed 2022-03-29 by ronagyl