analogdevicesinc/hdl #916: [BUG] AD9361 Xilinx axi_ad9361_lvds_if

AlexLao512, 2022-04-09:
Seems commit 83d3bded6312449766385730ab07b3afde1adbfe from years ago completely breaks RX on the AD9361. Prior commits of this file works perfectly for us, but the changes in this commit breaks it. When I look at the rx_frame signals in a logic analyze at runtime, the case statements for delineation can not possibly work. This is an RTL bug, tool and software versions should not matter here.

[image]

Used Software: No-OS
Tool version: Vivado 2019.1.3
HDL Release version hdl_2019_r2
Software Release version 2019_r2

dmitrii-galantsev, 2022-09-11:
Affirmative, can observe in my setup as well. 👎

johnathan-convertino-afrl, 2023-12-21:
Change lines 222 to 229 in hdl/library/axi_ad9361/xilinx
/axi_ad9361_lvds_if.v to the snippet below.

```verilog
always @(posedge l_clk) begin
    rx_error_r1 <= ~(({rx_frame, rx_frame_s} == 4'b1100) || ({rx_frame, rx_frame_s} == 4'b0011));
    rx_error_r2 <= ~(({rx_frame, rx_frame_s} == 4'b1111) || ({rx_frame, rx_frame_s} == 4'b1100) ||
                     ({rx_frame, rx_frame_s} == 4'b0000) || ({rx_frame, rx_frame_s} == 4'b0011));
  end

  always @(posedge l_clk) begin
      case ({rx_r1_mode, rx_frame, rx_frame_s})
```

A few years back I ran into this issue and fixed it for my own HDL project. The original code using rx_frame_s only makes zero sense, as its only a 2 bit vector being compared to a 4 bit vector. Fix is based on the previous code. I also changed the order of the case statement  as well. 

Wish I had documented the changes better, my commits don't reflect the reasons. It was bundled with other changes involved in arria10 timing. Lesson, make better commits.

IuliaCMoldovan, 2024-03-20:
Hi all,

Thank you for letting us know, and apologies for the very late reply; somehow we missed this. We don't really track the GitHub issues, as most of the inquiries and comments come through our [EngineerZone](https://ez.analog.com/fpga/f/q-a) forum.
This will be fixed in the future commits.

Best regards,
Iulia

PR #1460 (2024-11-27): axi_ad9361: Fixup LVDS RX_FRAME (#916)
## PR Description

The rx_error_r* on the axi_ad9361/xilinx/axi_ad9361_lvds_if.v were not implemented properly.
the author intended it to be the inverse of all valid frames combinations {rx_frame, rx_frame_s} but only compared against rx_frame_s.

Checking against all valid frames is also unnecessary, since checking for unequal DDR output values have the same practical effect.
It is also worth noting that at 2RX mode, a frame fulfills the 2 channels in 4 clock cycles, while  {rx_frame, rx_frame_s} looks back only 2 clock cycles.

[image]
UG-570, p95

This [... cut]