analogdevicesinc/hdl #930: [BUG] Latches in JESD204 receive peripheral

Prandr, 2022-05-02:
**Describe the bug**
Latches show up in the hierarchy of the JESD204 Receive peripheral  (submodule jesd204_rx_lane) when viewed in elaborated design in Vivado.  

**Screenshots**

[image]

**Desktop:**
Vivado 2021.2
hdl_2019_r2

ronagyl, 2022-05-02:
Can you observe any latches in the implemented design ? 

Thank you,
Laszlo

Prandr, 2022-05-03:
No, can't find them in the implemented design. But still abnormal to have them and I have a hunch about the cause.

in file jesd204_rx_lane.v

```Verilog
genvar i;
generate

for (i = 0; i < DATA_PATH_WIDTH; i = i + 1) begin: gen_char
  assign char[i] = phy_data[i*8+7:i*8];
  assign char_is_valid[i] = ~(phy_notintable[i] | phy_disperr[i]);

  always @(*) begin
    char_is_error[i] = ~char_is_valid[i];

    char_is_cgs[i] = 1'b0;
    charisk28[i] = 1'b0;
    unexpected_char[i] = 1'b0;

    if (phy_charisk[i] == 1'b1 && char_is_valid[i] == 1'b1) begin
      if (char[i][4:0] == 'd28) begin
        charisk28[i] = 1'b1;
        if (char[i][7:5] == 'd5) begin
          char_is_cgs[i] = 1'b1;
        end
      end else begin
        unexpected_char[i] = 1'b1;
      end
    end
  end
end
endgenerate
```
Perhaps the loop could be moved into the process instead of generating 4 processes.
I am a VHDL coder, so my understanding of Verilog is superficial.

ronagyl, 2022-05-03:
With Vivado 2021.1 (that hdl_2019_r2 requires) I could not reproduce the issue.  If you don't see latches in the implemented design I think is safe to use the design as is however I recommend using Vivado 2021.1 

We will address this issue once we upgrade the tool versions. 
Thank you for the heads up.

[image]

closed 2022-05-03 by ronagyl

Prandr, 2022-05-03:
Did you run OOC synthesis before opening elaborated design?
You need to use global synthesis then to see it.

ronagyl, 2022-05-03:
Elaboration is a step before of the synthesis and should not depend on it, but indeed the system block design output generation was set to OOC. 
Once I changed the output generation to Global and re-ran the elaboration the 'latches' appeared but after running the synthesis there are gone. I consider the design safe at the moment, rather it looks more like an elaboration issue since also inside Xilinx IPs I can find similar elements.