analogdevicesinc/hdl #990: [BUG] Possible bug in request_arb.v of axi_dmac core

catkira, 2022-08-18:
I am looking at this function and cant make sense of it:

```verilog
function compare_id;
  input [ID_WIDTH-1:0] a;
  input [ID_WIDTH-1:0] b;
  begin
    compare_id = a[ID_WIDTH-1] == b[ID_WIDTH-1];
    if (ID_WIDTH >= 2) begin
      if (a[ID_WIDTH-2] == b[ID_WIDTH-2]) begin
        compare_id = 1'b1;
      end
    end
    if (ID_WIDTH >= 3) begin
      if (a[ID_WIDTH-3:0] != b[ID_WIDTH-3:0]) begin
        compare_id = 1'b1;
      end
    end
  end
endfunction
```

Why is it comparing in some cases "==" and in another case "!="?
That numbers that are compared are gray coded, maybe that helps to figure out the meaning of this function...

larsclausen, 2022-08-18:
It's standard logic for comparing gray code FIFO addresses.

- https://zipcpu.com/blog/2018/07/06/afifo.html
- http://www.sunburst-design.com/papers/CummingsSNUG2002SJ_FIFO1.pdf

To understand this maybe going through a simple example will help. Imagine a FIFO with 4 entries. The FIFO can be in 5 different states (0, 1, 2, 3, 4 elements used). So you need 3 bits. Draw the table of all 3 bit gray codes in the right order and then look at it to see the pattern when FIFO should be considered full.

catkira, 2022-08-19:
thanks, that helped a lot!

closed 2022-08-19 by catkira