analogdevicesinc/hdl #1009: Using the IIO scope with KC705 and fmcdaq2

yashwanthresham, 2022-09-22:
I am using kc705 and fmcdaq2. Can I be able to see the dac output in IIO scope . If so, does it require any additional modules in the design?

AndreiGrozav, 2022-09-22:
You can see the output of the dac by capturing the data with the adc. Simply do a loop back with an sma cable.
Everything is there.

yashwanthresham, 2022-09-22:
Yes. But IIO shows No function found for the context. Also the device is physically connected the system through JTAG and UART.  Can you share the steps to be followed after loading the FPGA through SDK .

AndreiGrozav, 2022-09-22:
https://wiki.analog.com/resources/eval/user-guides/ad-fmcdaq2-ebz/software/linux/microblaze
enter ifconfig in the UART terminal.
In the newer versions of IIO-Osc, enter in the manual tab the ip you got from the UART as "ip:xxx.xxx.xxx.xxx". Then clock o refresh. If you don't get any devices, this could take more time and you should post this problem in Engineer Zone. FPGA reference designs forum. Is a good chance you will find a solution on the forum in one of the threads.