analogdevicesinc/hdl #1065: Problem with building of vivado project in vivado 2021.1[BUG]

sdrlab, 2023-01-19:
Hello , 
i tried to build adrv9364z7020 project using cygwin as follows:

### First step :
in cygwin export the follows in this way : 
[image]

### Second step : 
in cygwin i sourced vivado using : 
[image]

### Third step : 
after getting to my target path call the make function: 
[image]

but as you can see i got error : 

tac: cannot open `\'' for reading: No such file or directory
lib was unexpected at this time.
make: *** [../../scripts/project-xilinx.mk:117: lib] Error 255

what is the problem and how i can solve it ?
i will appriceate your help.
with best regards.
Dror

IstvanZsSzekely, 2023-01-19:
Hi Dror,

I tried building the project in a similar manner as you did. 
[image]
Please note that sourcing the settings64.sh in Cygwin under Windows OS will add the paths in Windows specific format, using '\\', instead of '/'. This will break the currently open Cygwin terminal's Path. Open a new terminal and at the second step add the path like this instead of sourcing it: 

export PATH=$PATH:/cygdrive/d/Xilinx/Vivado/202x.x/bin:/cygdrive/d/Xilinx/Vitis/202x.x/bin

Please make sure that the path is correct before running the command and take a look at [Building HDL](https://wiki.analog.com/resources/fpga/docs/build) for additional information. 

Best regards,
Istvan

sdrlab, 2023-01-19:
Hello Istavan , now i did the export line like you ask me to do (please see the following picture) but unfourtantly i got the following error now : 
[image]

And if i check the log file they mention (axi_ad9361_ip.log) we can see the following : 
`
****** Vivado v2021.1 (64-bit)
  **** SW Build 3247384 on Thu Jun 10 19:36:33 MDT 2021
  **** IP Build 3246043 on Fri Jun 11 00:30:35 MDT 2021
    ** Copyright 1986-2021 Xilinx, Inc. All Rights Reserved.

source axi_ad9361_ip.tcl
# source ../../scripts/adi_env.tcl
## set ad_hdl_dir [file normalize [file join [file dirname [info script]] "../"]]
## if [info exists ::env(ADI_HDL_DIR)] {
##   set ad_hdl_dir [file normalize $::env(ADI_HDL_DIR)]
## }
## if [info exists ::env(ADI_GHDL_DIR)] {
##   set ad_ghdl_dir [file normalize $::env(ADI_GHDL_DIR)]
## }
## set required_vivado_version "2021.2"
## if {[info exists ::env(REQUIRED_VIVADO_VERSION)]} {
##   set required_vivado_version $::env(REQUIRED_VIVADO_VERSION)
## } elseif {[info exists REQUIRED_VIVADO_VERSION]} {
##   set required_vivado_version $REQUIRED_VIVADO_VERSION
## }
## if {[info exists ::env(ADI_IGNORE_VERSION_CHECK)]} {
##   set IGNORE_VERSION_CHECK 1
## } elseif {![info exists IGNORE_VERSION_CHECK]} {
##   set IGNORE_VERSION_CHECK 0
## }
## if {![info exists REQUIRED_QUARTUS_VERSION]} {
##   set REQUIRED_QUARTUS_VERSION "21.4.0"
## }
## proc get_env_param {name default_value} {
##   if [info exists ::env($name)] {
##     puts "Getting from environment the parameter: $name=$::env($name) "
##     return $::env($name)
##   } else {
##     return $default_value
##   }
## }
# source $ad_hdl_dir/library/scripts/adi_ip_xilinx.tcl
## source $ad_hdl_dir/library/scripts/adi_xilinx_device_info_enc.tcl
### variable auto_set_param_list
### variable auto_set_param_list_overwritable
### variable fpga_series_list
### variable fpga_family_list
### variable speed_grade_list
### variable dev_package_list
### variable xcvr_type_list
### variable fpga_voltage_list
### set auto_set_param_list { \
###           DEV_PACKAGE \
###           SPEED_GRADE \
###           FPGA_FAMILY \
###           FPGA_TECHNOLOGY }
### set auto_set_param_list_overwritable { \
###           FPGA_VOLTAGE \
###           XCVR_TYPE }
### set fpga_technology_list { \
###         { Unknown     0 } \
###         { 7series     1 } \
###         { ultrascale  2 } \
###         { ultrascale+ 3 } \
###         { versal      4 }}
### set fpga_family_list { \
###         { Unknown      0 } \
###         { artix        1 } \
###         { kintex       2 } \
###         { virtex       3 } \
[... 1261 lines cut ...]
## set_msg_config -id {IP_Flow 19-3656} -new_severity INFO
## set_msg_config -id {IP_Flow 19-4623} -new_severity INFO
## set_msg_config -id {IP_Flow 19-459} -new_severity INFO
## set_msg_config -id {Synth 8-3331} -new_severity INFO
## set_msg_config -id {Synth 8-2490} -new_severity WARNING
## set_msg_config -id {Designutils 20-3303} -string "HDPYFinalizeIO" -new_severity INFO
## set_msg_config -id {Place 30-73} -string "axi_spi" -new_severity WARNING
## set_msg_config -string "PCW_UIPARAM_DDR_DQS_TO_CLK_DELAY" -new_severity WARNING
INFO: [IP_Flow 19-234] Refreshing IP repositories
INFO: [IP_Flow 19-1700] Loaded user IP repository 'c:/adi/hdl/library'.
INFO: [IP_Flow 19-3656] If you move the project, the path for repository 'c:/adi/hdl/library' may become invalid. A better location for the repository would be in a path adjacent to the project. (Current project location is 'c:/adi/hdl/library/util_tdd_sync'.)
WARNING: [IP_Flow 19-2248] Failed to load user IP repository 'c:/adi/ghdl/library'; Can't find the specified path.
If this directory should no longer be in your list of user repositories, go to the IP Settings dialog and remove it.
INFO: [IP_Flow 19-2313] Loaded Vivado IP repository 'C:/Xilinx/Vivado/2021.1/data/ip'.
# adi_ip_files util_tdd_sync [list \
#   "$ad_hdl_dir/library/common/util_pulse_gen.v" \
#   "util_tdd_sync_constr.xdc" \
#   "util_tdd_sync.v"]
# adi_ip_properties_lite util_tdd_sync
ERROR: [Common 17-217] Failed to load feature 'ipservices'.
INFO: [Common 17-206] Exiting Vivado at Thu Jan 19 13:10:18 2023...
`
I don't understand what is the ipservices it couldn't load .

what is the problem now ?

AndreiGrozav, 2023-01-19:
We can't reproduce this.
Go to library/util_tdd_sync and do:
make clean
make

closed 2023-03-29 by IuliaCMoldovan