analogdevicesinc/hdl #1084: hdl/projects/adrv9361z7035/readme specifies non-existent folder

Danevolent, 2023-02-09:
Building, Generating Elf Files (easy & efficient method)
specifies 
[some-directory]> make -C no-OS/adrv9361z7035/ccbob_cmos
to build Elf files.

The folder "adrv9361z7035" does not exist.
Not in the no-OS folder, and ALSO not in the no-OS/projects/ folder.

projects/AD9361 is the closet description match.

buha, 2023-02-09:
Hi @Danevolent ,
try building no-OS with this flag enabled:
https://github.com/analogdevicesinc/no-OS/blob/master/projects/ad9361/src/app_config.h#L53

Danevolent, 2023-02-10:
Thank you, although I think it's a problem on my end.
When I attempted to build the no-OS with that flag enabled (while following directions: https://wiki.analog.com/resources/no-os/build)
I got the following error:
[image]

buha, 2024-02-05:
Hey @TachiOni , is this still an issue for you ? There were changes in the build process in the meantime.
Sorry for missing on your follow-up reply, I mostly monitor EngineerZone forum.

closed 2024-03-20 by IuliaCMoldovan