analogdevicesinc/hdl #1147: Support for AD2S1210 evaluation board on ZedBoard

pamolloy, 2023-07-18:
EVAL-AD2S1210 -> SDP-I-FMC Interposer -> ZedBoard

mhennerich, 2023-07-21:
Files got merged with this PR:  https://github.com/adi-ses/sea-misc/pull/10

acostina, 2023-07-21:
working branch is at: https://github.com/analogdevicesinc/hdl/tree/dev_ad2s1210_sdz_zed

Skallwar, 2023-07-24:
Boot bin: https://github.com/adi-ses/sea-misc/tree/main/AD2S1210-PRJ/hdl/boot

Skallwar, 2023-07-24:
Changing the BOOT.bin with this [one](https://github.com/adi-ses/sea-misc/blob/main/AD2S1210-PRJ/hdl/boot/BOOT.BIN) lead to kernel panic:
```
Console: colour dummy device 80x30
Calibrating delay loop (skipped), value calculated using timer frequency.. 666.66 BogoMIPS (lpj=3333333)
pid_max: default: 32768 minimum: 301
Mount-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
CPU: Testing write buffer coherency: ok
CPU0: Spectre v2: using BPIALL workaround
CPU0: thread -1, cpu 0, socket 0, mpidr 80000000
Setting up static identity map for 0x100000 - 0x100060
rcu: Hierarchical SRCU implementation.
smp: Bringing up secondary CPUs ...
CPU1: thread -1, cpu 1, socket 0, mpidr 80000001
CPU1: Spectre v2: using BPIALL workaround
smp: Brought up 1 node, 2 CPUs
SMP: Total of 2 processors activated (1333.33 BogoMIPS).
CPU: All CPU(s) started in SVC mode.
devtmpfs: initialized
VFP support v0.3: implementor 41 architecture 3 part 30 variant 9 rev 4
clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
futex hash table entries: 512 (order: 3, 32768 bytes, linear)
pinctrl core: initialized pinctrl subsystem
NET: Registered protocol family 16
DMA: preallocated 256 KiB pool for atomic coherent allocations
thermal_sys: Registered thermal governor 'step_wise'
hw-breakpoint: found 5 (+1 reserved) breakpoint and 1 watchpoint registers.
hw-breakpoint: maximum watchpoint size is 4 bytes.
zynq-ocm f800c000.ocmc: ZYNQ OCM pool: 256 KiB @ 0x(ptrval)
e0001000.serial: ttyPS0 at MMIO 0xe0001000 (irq = 33, base_baud = 3125000) is a xuartps
printk: console [ttyPS0] enabled
SCSI subsystem initialized
usbcore: registered new interface driver usbfs
usbcore: registered new interface driver hub
usbcore: registered new device driver usb
mc: Linux media interface: v0.10
videodev: Linux video capture interface: v2.00
jesd204: found 0 devices and 0 topologies
FPGA manager framework
Advanced Linux Sound Architecture Driver Initialized.
clocksource: Switched to clocksource arm_global_timer
NET: Registered protocol family 2
tcp_listen_portaddr_hash hash table entries: 512 (order: 0, 6144 bytes, linear)
TCP established hash table entries: 4096 (order: 2, 16384 bytes, linear)
TCP bind hash table entries: 4096 (order: 3, 32768 bytes, linear)
TCP: Hash tables configured (established 4096 bind 4096)
UDP hash table entries: 256 (order: 1, 8192 bytes, linear)
UDP-Lite hash table entries: 256 (order: 1, 8192 bytes, linear)
NET: Registered protocol family 1
RPC: Registered named UNIX socket transport module.
RPC: Registered udp transport module.
RPC: Registered tcp transport module.
RPC: Registered tcp NFSv4.1 backchannel transport module.
hw perfevents: no interrupt-affinity property for /pmu@f8891000, guessing.
hw perfevents: enabled with armv7_cortex_a9 PMU driver, 7 counters available
workingset: timestamp_bits=30 max_order=17 bucket_order=0
NFS: Registering the id_resolver key type
Key type id_resolver registered
Key type id_legacy registered
nfs4filelayout_init: NFSv4 File Layout Driver Registering...
fuse: init (API version 7.32)
io scheduler mq-deadline registered
io scheduler kyber registered
zynq-pinctrl 700.pinctrl: zynq pinctrl initialized
8<--- cut here ---
Unhandled fault: imprecise external abort (0x406) at 0x00000000
pgd = (ptrval)
[00000000] *pgd=00000000
Internal error: : 406 [#1] PREEMPT SMP ARM
Modules linked in:
[... 45 lines cut ...]
bfe0: 00000000 00000000 00000000 00000000 00000013 00000000
Code: e5933000 e1a02003 e58d300c f57ff04f (e3a03e36)
---[ end trace 506a2718bbc67349 ]---
Kernel panic - not syncing: Attempted to kill init! exitcode=0x0000000b
CPU1: stopping
CPU: 1 PID: 0 Comm: swapper/1 Tainted: G      D           5.10.0-98744-g64e7a607b659 #1788
Hardware name: Xilinx Zynq Platform
[<c010da8c>] (unwind_backtrace) from [<c010a164>] (show_stack+0x10/0x14)
[<c010a164>] (show_stack) from [<c0b29dbc>] (dump_stack+0xc0/0xd4)
[<c0b29dbc>] (dump_stack) from [<c010bf58>] (do_handle_IPI+0x120/0x148)
[<c010bf58>] (do_handle_IPI) from [<c010bf98>] (ipi_handler+0x18/0x20)
[<c010bf98>] (ipi_handler) from [<c016f080>] (handle_percpu_devid_irq+0x78/0x150)
[<c016f080>] (handle_percpu_devid_irq) from [<c0169240>] (__handle_domain_irq+0x7c/0xd0)
[<c0169240>] (__handle_domain_irq) from [<c043b3d4>] (gic_handle_irq+0x80/0x94)
[<c043b3d4>] (gic_handle_irq) from [<c0100b0c>] (__irq_svc+0x6c/0xa8)
Exception stack(0xc1869f60 to 0xc1869fa8)
9f60: 00000000 00000d3c dfba0184 c0117280 c1868000 00000001 c1503f14 c1503f50
9f80: 00000000 c144cc10 00000000 c1869fb8 c1898040 c1869fb0 c01075c4 c01075c8
9fa0: 60000013 ffffffff
[<c0100b0c>] (__irq_svc) from [<c01075c8>] (arch_cpu_idle+0x38/0x3c)
[<c01075c8>] (arch_cpu_idle) from [<c014ab8c>] (do_idle+0x204/0x294)
[<c014ab8c>] (do_idle) from [<c014af1c>] (cpu_startup_entry+0x18/0x1c)
[<c014af1c>] (cpu_startup_entry) from [<0010156c>] (0x10156c)
---[ end Kernel panic - not syncing: Attempted to kill init! exitcode=0x0000000b ]---
```

mhennerich, 2023-07-24:
can you explain exactly what you did?

Skallwar, 2023-07-25:
I swapped the BOOT.bin with the one you provided. We didn't change the devicetree but from what I gather BOOT.bin and devicetree need to be align.

Blocked by: adi-ses/sea-misc/issues/21

pamolloy, 2023-08-15:
We were able to boot and interface with the AD2S1210 using the new `boot.bin`

closed 2023-08-17 by pamolloy