analogdevicesinc/hdl #1170: util_axis_fifo: m_axis_tkeep & m_axis_tlast does not have driver (Vivado warning)

StephanAAU, 2023-09-08:
When using util_axis_fifo and having TKEEP_EN & TLAST_EN set to zero, Vivado will throw a warning (Synth 8-3848) that they do not have a driver.

I suggest setting them to both to '0' at line 297 in util_axis_fifo.v such that they are driven both to avoid the warning and to ensure known behavior.
[https://github.com/analogdevicesinc/hdl/blob/91ec36f4179c5c8b91e619c589790f1437955125/library/util_axis_fifo/util_axis_fifo.v#L297
](https://github.com/analogdevicesinc/hdl/blob/91ec36f4179c5c8b91e619c589790f1437955125/library/util_axis_fifo/util_axis_fifo.v#L297)

Warning message:
>WARNING: [Synth 8-3848] Net m_axis_tkeep in module/entity util_axis_fifo does not have driver. [.../util_axis_fifo.v:53]
>WARNING: [Synth 8-3848] Net m_axis_tlast in module/entity util_axis_fifo does not have driver. [.../util_axis_fifo.v:54]

StephanAAU, 2023-09-15:
Additional comment to the suggestion: Should also be added for the other configurations of the axis_fifo

IstvanZsSzekely, 2024-01-22:
Hi @StephanAAU ,

Thank you for giving us a heads-up, and sorry for this very late response. 
I'll take a look at it, and will keep you updated with the changes. 

Regards,
Istvan

IstvanZsSzekely, 2024-01-22:
Created a [PR](https://github.com/analogdevicesinc/hdl/pull/1255) regarding your observations. I hard wired tkeep to 1 instead of 0, since we want the data to be processed. I also updated the other configurations as you suggested. 

Regards,
Istvan

closed 2024-01-22 by IstvanZsSzekely

StephanAAU, 2024-01-22:
Cool - great work :+1: