analogdevicesinc/hdl #1237: [BUG] ad9081_fmca_ebz on VCU128 on main branch fails to build with vivado 2023.1

pgsys, 2023-12-13:
I cannot build the ad9081_fmca_ebz project with the VCU128 target.  When I try, I get an error related to the HBM memory size.  I tried changing the LENGTH_WIDTH parameter but it had no effect, it appears the memory size is calculated and passed in from another parameter which I'm unclear about (I'm a TCL novice)

**To Reproduce**
1. Use Vivado 2023.1 on Ubuntu Linux 22.04
2. clone adi HDL repo
3. in the ad9081_fmca_ebz/vcu128 subdirectory run `make`

Here is the vivado log dump:
ERROR: [IP_Flow 19-3452] Invalid long/float value '' specified for parameter 'HBM sections per master(HBM_SEGMENTS_PER_MASTER)' for BD Cell 'storage_unit'.
ERROR: [IP_Flow 19-3461] Value '24' is out of the range for parameter 'Storage size(LENGTH_WIDTH)' for BD Cell 'storage_unit' . Valid values are - 28, 29, 30, 31, 32, 33, 34
CRITICAL WARNING: [IP_Flow 19-5622] Failed to create IP instance 'system_storage_unit_0'. Failed to customize IP instance 'system_storage_unit_0'. Failed to load customization data
ERROR: [BD 41-1712] Create IP failed with errors
ERROR: [BD 5-7] Error: running create_bd_cell  -vlnv analog.com:user:util_hbm:1.0 -type ip -name storage_unit .
ERROR: [Common 17-39] 'create_bd_cell' failed due to earlier errors.

I am trying to build from the latest main branch of the HDL repo.

IuliaCMoldovan, 2023-12-15:
Hi @pgsys,

Thank you for letting us know.
I'll check this issue and come back as soon as I have an answer.

Iulia

IuliaCMoldovan, 2023-12-21:
Hi,

Sorry for the delay; I was able to reproduce this.
I'll start investigating but only in January I can come back with an answer.

Regards,
Iulia

pgsys, 2024-01-24:
Iulia,  
I see you are actively working on a branch for this, thank you very much.  Let me know if there's some way I can help.

IuliaCMoldovan, 2024-01-30:
Hi,

Could you try building the project from vivado23_2 branch? 

This is solved by fixing a mistake in the util_hbm IP and changing something else regarding the syntax of a expression (which it seems that 2023.1 version didn't like). 
But then, the project fails with timing constraints not met, in the Vivado 2023.1 version. This is rather strange, as in the previous versions it worked, but it appears that this version has some other measurements.
Since we're in the process of updating to the newer Vivado version (2023.2), there's no point in fixing the timing issue in 2023.1.
Thus, building the project with 2023.2 and with the util_hbm IP changes, fixes this project.

Iulia

closed 2024-02-13 by IuliaCMoldovan