analogdevicesinc/hdl #1250: [BUG]

northshaur, 2024-01-16:
**Describe the bug**
Attempting to run make in: /hdl/projects/ad9081_fmca_ebz/vcu118

getting the following error, it may be because of the version mismatch but I suspect its something else due to this thread (https://ez.analog.com/fpga/f/q-a/571950/hdl-build-fail-for-adrv9009-with-zcu102):
```
connect_bd_net /rx_mxfe_tpl_core/adc_dovf /rx_mxfe_tpl_core/adc_tpl_core/adc_dovf
ERROR: [IP_Flow 19-3452] Invalid long/float value '"0"' specified for parameter 'Mem Type(MEM_TYPE)' for BD Cell 'i_data_offload'.
CRITICAL WARNING: [IP_Flow 19-5622] Failed to create IP instance 'system_i_data_offload_0'. Failed to customize IP instance 'system_i_data_offload_0'. Failed to load customization data
ERROR: [BD 41-1712] Create IP failed with errors
ERROR: [BD 5-7] Error: running create_bd_cell  -vlnv analog.com:user:data_offload:1.0 -type ip -name i_data_offload .
ERROR: [Common 17-39] 'create_bd_cell' failed due to earlier errors.

    while executing
"create_bd_cell -type ip -vlnv [get_ipdefs -all -filter "VLNV =~ *:${i_ip}:* &&  design_tool_contexts =~ *IPI* && UPGRADE_VERSIONS == \"\""] ${i_name}"
    (procedure "ad_ip_instance" line 3)
    invoked from within
"ad_ip_instance data_offload i_data_offload [list  MEM_TYPE [expr $mem_type == 0 ? 0 : 1]  MEM_SIZE_LOG2 [log2 $mem_size]  TX_OR_RXN_PATH $datapath_typ..."
    (procedure "ad_data_offload_create" line 36)
    invoked from within
"ad_data_offload_create $adc_data_offload_name  0  $adc_do_mem_type  $adc_data_offload_size  $adc_data_width  $adc_data_width  $do_axi_data_width  $SHA..."
    invoked from within
"if {$INTF_CFG != "TX"} {
  create_bd_port -dir I rx_device_clk

  if {$ADI_PHY_SEL == 1} {
    ad_ip_instance axi_adxcvr axi_mxfe_rx_xcvr
    ad_ip_pa..."
    (file "/home/shaur/development/eval_boards/hdl/projects/ad9081_fmca_ebz/common/ad9081_fmca_ebz_bd.tcl" line 197)
```

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '/hdl/projects/ad9081_fmca_ebz/vcu118'
2. run make
4. See error

**Expected behavior**
Make to finish cleanly 

a
**Desktop (please complete the following information):**
 - Project name and used carrier board: ad9081_fmca_ebz/vcu118
 - Used Software: [e.g. Linux or No-OS]: Linux
 - Tool version [e.g. Vivado 2018.3, Quartus PRO 18.1]: Vivado 23.2
 - HDL Release version [e.g. hdl_2019_r1]L HDL 2022 r2
 - Software Release version [e.g. 2019_R1]

**Additional context**
Add any other context about the problem here.

tfcollins, 2024-01-17:
Release 2022-R2 supports only Vivado 2022.2. Please use that tool. If the build still fails please post the full build log.

northshaur, 2024-01-17:
Hello, ty for the feedback! I tried it again and got a  failure in bitstream generation 
[timing_impl.log](https://github.com/analogdevicesinc/hdl/files/13968532/timing_impl.log)

northshaur, 2024-01-17:
It appears it needs a license file for bd_4bad_mac_0_core,

tfcollins, 2024-01-17:
To debug we need the build log not timing reports. Make should tell you what this file is when the build finishes.

northshaur, 2024-01-17:
Apologies, I attached the wrong log. I got the eval license for the temac core and ran again, and got the following error 
[ad9081_fmca_ebz_vcu118_vivado.log](https://github.com/analogdevicesinc/hdl/files/13969047/ad9081_fmca_ebz_vcu118_vivado.log)

for some reason the license is still not found so I am investigating that still

northshaur, 2024-01-17:
Sorry for the back and forth - it Built, AD is using some IP cores that require licensing (paid) and that was the issue. Any idea if thats expected for these projects?

tfcollins, 2024-01-17:
We use the AMD Ethernet IP in the Virtex/Kintex designs. Primarily to control and stream data off the board. You can probably get an eval license from AMD for that core if you just want to use it for testing or a proof of concept.

But yes this is expected.

northshaur, 2024-01-17:
Got it, thank you! sorry for the trouble

closed 2024-01-29 by IuliaCMoldovan