analogdevicesinc/hdl #1254: [BUG] "axi_pwm_gen: Offset mecanism fix" breaks pwm output

dlech, 2024-01-20:
**Describe the bug**
The PWM output in `projects/ad738x_fmc/zed` (and likely others) is not working using the current `main`. 

**To Reproduce**
Steps to reproduce the behavior:
1. Build `projects/ad738x_fmc/zed` from `main` branch (d300b9c55c6d8a1a2233d46d1b6f42274ecf2e33)
2. Make sure your Linux pwmgen driver is updated for pwmgen v2.
3. Try to use a previously known working branch for your Linux kernel (with modifications from 2.) 
4. SPI Engine offload is never triggered by the PWM.

**Expected behavior**
The PWM should trigger the SPI Engine offload.

**Desktop (please complete the following information):**
 - Project name and used carrier board [e.g. FMCOMMS2 ZC706]

   `projects/ad738x_fmc/zed`  with EVAL-AD7383FMC + Zed

 - Used Software: [e.g. Linux or No-OS]

    Linux

 - Tool version [e.g. Vivado 2018.3, Quartus PRO 18.1]

   Vivado v2023.1 (64-bit)

 - HDL Release version [e.g. hdl_2019_r1]

    d300b9c55c6d8a1a2233d46d1b6f42274ecf2e33

 - Software Release version [e.g. 2019_R1]

    mainline kernel

**Additional context**

Reverting commit https://github.com/analogdevicesinc/hdl/commit/e7dd5ce3948440e2cf2eb3f509251ae739c32e85 "axi_pwm_gen: Offset mecanism fix" fixes the issue. So this is the likely cause of the break.

acostina, 2024-01-29:
I don't seem to find the Linux branch you are using. Can you share it ?

PopPaul2021, 2024-02-01:
Hi @dlech, we attempted to replicate the configuration you suggested using the branch you directed us to. It appears that there is no defconfig file for the Zynq FPGA. When we used the one from the main branch and created an uImage, the devices were not probed. Could you describe us the steps you take when generating the uImage from the dlech/spi-engine-offload-ad7980-pwmgen-v2 branch? Thanks!

dlech, 2024-02-01:
There are a few modules the probably need to be enabled (e.g. `CONFIG_AD7380`, `CONFIG_IIO_PWM_TRIGGERED_DMA_BUFFER`). 

Try this config file: [ad7380-config.txt](https://github.com/analogdevicesinc/hdl/files/14127507/ad7380-config.txt)

PIoandan, 2024-02-02:
Hello @dlech. We tried to get uImage and devicetree with ad7380-config.txt file, but we got this error:
[image]
[image]
[image]
Can you share the uImage and devicetree.dtb files and even the BOOT.BIN ?

dlech, 2024-02-02:
Oof, weird error. Here are the requested files. 

[hdl-issue-1254.zip](https://github.com/analogdevicesinc/hdl/files/14145436/hdl-issue-1254.zip)

FYI, I pushed one more commit to the Linux branch to fix an issue that only showed up when compiling everything as builtin instead of .ko modules. That fix is included in the uImage attached.

PIoandan, 2024-02-07:
Hello @dlech. Here is an archive with the BOOT.BIN files for the one/two wire mode version after an error in hdl was identified and fixed. 
[BOOT_BIN_AD738x.zip](https://github.com/analogdevicesinc/hdl/files/14191084/BOOT_BIN_AD738x.zip)
The modification can be find on the main branch: https://github.com/analogdevicesinc/hdl/blob/main/library/axi_pwm_gen/axi_pwm_gen.sv#L301.
Here is the signal that was obtained after I tested on hardware, when I applied signals on the both channels for 1 wire mode version.
[image]

PIoandan, 2024-03-20:
Hello @dlech. Did the boot files I provided fix the problem? I ask this because I want to know if this issue can be closed or not. Thanks!

dlech, 2024-03-20:
I haven't had a chance to test with AD783x yet since I've been working on AD7944 instead.

But we did test a different BOOT.BIN built from the main branch that connects the PWM outputs directly to the PMOD header and it is working.

So, I think we can go ahead and close this.