analogdevicesinc/hdl #1389: [BUG] Possible off-by-one in SPI Engine delay

dlech, 2024-07-23:
**Describe the bug**
We are working on a Linux Driver for AD738x. While troubleshooting an issue with the driver, we observed that the measured delay of a SPI Engine delay instruction is less than the requested delay.

We have a SPI message that does a CS :arrow_down:, delay, CS :arrow_up: :arrow_down:, delay, CS :arrow_up: :arrow_down: read, CS :arrow_up: :arrow_down:, read, CS :arrow_up:.

This results in the following SPI engine instruction sequence:

```
[   49.695838] ad7380 spi0.0: cmd[0]: 0x2102 CPOL=1, CPHA=0
[   49.695853] ad7380 spi0.0: cmd[1]: 0x10fe assert CS0
[   49.695866] ad7380 spi0.0: cmd[2]: 0x2007 CLKDIV=8 (10MHz SCLK)
[   49.695877] ad7380 spi0.0: cmd[3]: 0x3104 delay 5 ticks (500ns)
[   49.695888] ad7380 spi0.0: cmd[4]: 0x10ff deassert CS0
[   49.695899] ad7380 spi0.0: cmd[5]: 0x10fe assert CS0
[   49.695910] ad7380 spi0.0: cmd[6]: 0x3104 delay 5 ticks (500ns)
[   49.695921] ad7380 spi0.0: cmd[7]: 0x10ff deassert CS0
[   49.695932] ad7380 spi0.0: cmd[8]: 0x10fe assert CS0
[   49.695943] ad7380 spi0.0: cmd[9]: 0x2212 bits per word = 18
[   49.695954] ad7380 spi0.0: cmd[10]: 0x0201read xfer
[   49.695966] ad7380 spi0.0: cmd[11]: 0x10ff deassert CS0
[   49.695977] ad7380 spi0.0: cmd[12]: 0x10fe assert CS0
[   49.695989] ad7380 spi0.0: cmd[13]: 0x0201read xfer
[   49.696000] ad7380 spi0.0: cmd[14]: 0x10ff deassert CS0
[   49.696012] ad7380 spi0.0: cmd[15]: 0x2000 CLKDIV=0
[   49.696023] ad7380 spi0.0: cmd[16]: 0x3001 SYNC=1
```

**To Reproduce**
TODO: we can provide boot files and instructions if needed

**Expected behavior**
The second delay (green timing marker 1) should be >= the requested 500ns.

**Screenshots**

[image]

**Desktop (please complete the following information):**
 - Project name and used carrier board:  EVAL-AD7386FMCZ
 - Used Software: Linux
 - Tool version  Vivado 2023.2
 -  HDL Release version: commit 1652d0bf6
 - Software Release version: mainline kernel

**Additional context**

Since we are getting ~400 ns instead of ~500 ns and SCLK ticks are 100 ns, it seems like this could be an off-by-one error in the delay implementation.

dlech, 2024-07-23:
I also just built a new BOOT.BIN from the current main (commit 46aa12bbe) and the problem is the same.

LBFFilho, 2024-08-19:
Hi @dlech, I've created a testbench to look a bit more into this, you can find it here: https://github.com/analogdevicesinc/testbenches/tree/spi_delay
The test_delay test case does something similar to your test, with some small differences. Here's the relevant excerpt, where it writes the offload program:

```
  axi_write (`SPI_ENGINE_SPI_REGMAP_BA + GetAddrs(AXI_SPI_ENGINE_OFFLOAD0_CDM_FIFO), `INST_CFG);
  axi_write (`SPI_ENGINE_SPI_REGMAP_BA + GetAddrs(AXI_SPI_ENGINE_OFFLOAD0_CDM_FIFO), `INST_PRESCALE);
  axi_write (`SPI_ENGINE_SPI_REGMAP_BA + GetAddrs(AXI_SPI_ENGINE_OFFLOAD0_CDM_FIFO), `INST_DLENGTH);
  if (`CS_ACTIVE_HIGH) begin
    axi_write (`SPI_ENGINE_SPI_REGMAP_BA + GetAddrs(AXI_SPI_ENGINE_OFFLOAD0_CDM_FIFO), `SET_CS_INV_MASK(8'hFF));
  end
  axi_write (`SPI_ENGINE_SPI_REGMAP_BA + GetAddrs(AXI_SPI_ENGINE_OFFLOAD0_CDM_FIFO), `SET_CS(8'hFE));
  axi_write (`SPI_ENGINE_SPI_REGMAP_BA + GetAddrs(AXI_SPI_ENGINE_OFFLOAD0_CDM_FIFO), `SLEEP(4));
  axi_write (`SPI_ENGINE_SPI_REGMAP_BA + GetAddrs(AXI_SPI_ENGINE_OFFLOAD0_CDM_FIFO), `SET_CS(8'hFF));
  axi_write (`SPI_ENGINE_SPI_REGMAP_BA + GetAddrs(AXI_SPI_ENGINE_OFFLOAD0_CDM_FIFO), `SET_CS(8'hFE));
  axi_write (`SPI_ENGINE_SPI_REGMAP_BA + GetAddrs(AXI_SPI_ENGINE_OFFLOAD0_CDM_FIFO), `SLEEP(4));
  axi_write (`SPI_ENGINE_SPI_REGMAP_BA + GetAddrs(AXI_SPI_ENGINE_OFFLOAD0_CDM_FIFO), `SET_CS(8'hFF));
  axi_write (`SPI_ENGINE_SPI_REGMAP_BA + GetAddrs(AXI_SPI_ENGINE_OFFLOAD0_CDM_FIFO), `SET_CS(8'hFE));
  axi_write (`SPI_ENGINE_SPI_REGMAP_BA + GetAddrs(AXI_SPI_ENGINE_OFFLOAD0_CDM_FIFO), `SET_DLENGTH(18));
  axi_write (`SPI_ENGINE_SPI_REGMAP_BA + GetAddrs(AXI_SPI_ENGINE_OFFLOAD0_CDM_FIFO), `INST_RD);
  axi_write (`SPI_ENGINE_SPI_REGMAP_BA + GetAddrs(AXI_SPI_ENGINE_OFFLOAD0_CDM_FIFO), `SET_CS(8'hFF));
  axi_write (`SPI_ENGINE_SPI_REGMAP_BA + GetAddrs(AXI_SPI_ENGINE_OFFLOAD0_CDM_FIFO), `SET_CS(8'hFE));
  axi_write (`SPI_ENGINE_SPI_REGMAP_BA + GetAddrs(AXI_SPI_ENGINE_OFFLOAD0_CDM_FIFO), `INST_RD);
  axi_write (`SPI_ENGINE_SPI_REGMAP_BA + GetAddrs(AXI_SPI_ENGINE_OFFLOAD0_CDM_FIFO), `INST_SYNC | 1);
```

The largest difference is that the clock divider config is put before the cs change. The other thing is a bit symbolic: I'm considering the Sleep Instruction with parameter 4 as a delay of 4 ticks. [That's because the parameter for `t` for that instruction is not implicitly incremented, meaning it waits `t` ticks, not `t+1`](https://analogdevicesinc.github.io/hdl/library/spi_engine/instruction-format.html#sleep-instruction). This is in contrast with the prescaler register, which does that incrementing behavior.

All that to say: the behavior seems good in the testbench, it sleeps for about 400ns. And the green timing markers on your test seem correct to me as well.

dlech, 2024-08-19:
Thanks for looking into this. I wonder if this changed sometime in the last 9 years. The upstream Linux driver that was written 9 years ago expects t+1 ticks. And the [old wiki page](https://wiki.analog.com/resources/fpga/peripherals/spi_engine/instruction_format) also says the same.

[image]

It is only the [new docs page](https://analogdevicesinc.github.io/hdl/library/spi_engine/instruction-format.html#sleep-instruction) that says it should be t ticks.

[image]

So my question is: is the original docs and Linux implementation correct and there has been a regression in the HDL or was the original design intent what is currently implemented and has it really been wrong in the documentation and Linux kernel for 9 years and no one noticed until now?

LBFFilho, 2024-08-27:
I've opened a PR with the fix for this, also adds some historical context which explains why I was confused at first.

closed 2024-08-30 by LBFFilho