analogdevicesinc/hdl #1534: [BUG]

myjtag, 2024-12-02:
**Describe the bug**
I can not build any of the projects on win 11 and vivado 2024.2

**To Reproduce**
I'm using win11 and I have installed Cygwin64 and have followed the repo instructions.
I run Cygwin64 and then typed these commands
pip install pip --upgrade
cd docs ; pip install -r requirements.txt --upgrade
then this one
cd library ; make
the make would complain that there is no makefile
and this command has no makefile either
cd docs ; make html
then I try to set Windows environment setup with these commands

source /cygdrive/c/Xilinx/Vivado/2024.2/settings64.sh
export PATH=$PATH:/cygdrive/c/Xilinx/Vivado/2024.2/bin
export PATH=$PATH:/cygdrive/c/Xilinx/Vivado_HLS/2024.2/bin
export PATH=$PATH:/cygdrive/c/Xilinx/Vitis/2024.2/bin
export PATH=$PATH:/cygdrive/c/Xilinx/Vitis/2024.2/gnu/microblaze/nt/bin
export PATH=$PATH:/cygdrive/c/Xilinx/Vitis/2024.2/gnu/arm/nt/bin
export PATH=$PATH:/cygdrive/c/Xilinx/Vitis/2024.2/gnu/microblaze/linux_toolchain/nt64_be/bin
export PATH=$PATH:/cygdrive/c/Xilinx/Vitis/2024.2/gnu/microblaze/linux_toolchain/nt64_le/bin
export PATH=$PATH:/cygdrive/c/Xilinx/Vitis/2024.2/gnu/aarch32/nt/gcc-arm-none-eabi/bin

then I verify environment setup
[image]

then try to build some projects
[image]

what's going on wrong?

IuliaCMoldovan, 2024-12-02:
Hi,

You're running make in the wrong folder. You should go further down to a carrier from that folder.
https://analogdevicesinc.github.io/hdl/projects/ad7606x_fmc/index.html
Also, you need to run `export ADI_IGNORE_VERSION_CHECK=1` because we haven't yet migrated towards 2024.2, but we're still with Vivado 2023.2.

If you encounter other issues, please open a thread on our EngineerZone community and we'll assist you there: https://ez.analog.com/fpga/f/q-a

Best regards,
Iulia

closed 2024-12-02 by IuliaCMoldovan

myjtag, 2024-12-03:
thanks Iulia, I still got problems and can not build it and I explain it on engineering zone, But my answer was flagged as spam!!! can you check it there and help what's the problem.

Best Regards