analogdevicesinc/hdl #1596: Request for Bare-Metal Reference Project Source Code for FMCOMMS3 on Vitis

MehmettAkiff, 2025-03-03:
Hello,

I would like to run a bare-metal reference project for FMCOMMS3, but I need access to the source code for this project in Vitis. Unfortunately, I am not yet proficient enough in Vivado to manually write the necessary code for Vitis.

Could you provide the source code or direct me to the relevant repository where I can find the complete bare-metal reference project for FMCOMMS3 on Vitis? I checked the existing repositories, but I couldn't locate a straightforward implementation that I can use directly.

Thank you for your help!

Best regards,

IuliaCMoldovan, 2025-03-03:
Baremetal project source code: https://github.com/analogdevicesinc/no-OS/tree/main/projects/ad9361
And for support, open a question on https://ez.analog.com/microcontroller-no-os-drivers/ forum.

closed 2025-03-03 by IuliaCMoldovan

MehmettAkiff, 2025-03-03:
Thank you for your response and for sharing the repository. However, I am specifically looking for a complete bare-metal reference project that works out-of-the-box with the ZCU102 and FMCOMMS3 in Vitis, without requiring me to manually write or modify code.

The repository you shared contains only the AD9361 chip's drivers and related files, but I couldn't find a fully implemented example that directly runs on my hardware setup.

Could you please provide a direct link to a ready-to-use bare-metal reference project for ZCU102 + FMCOMMS3 in Vitis? Or, if such a project does not exist, could you guide me on how to properly integrate the provided drivers into a complete Vitis project?

Thank you for your help!

IuliaCMoldovan, 2025-03-03:
That project is used for all AD9361/AD9364-based evaluation boards, and it works out of the box, no need to change any code, just uncomment one line to specify that you're using a Xilinx platform: https://github.com/analogdevicesinc/no-OS/blob/main/projects/ad9361/src/app_config.h#L43
Project documentation: https://wiki.analog.com/resources/eval/user-guides/ad-fmcomms2-ebz/software/baremetal
How to build the no-OS application: https://wiki.analog.com/resources/no-os/build