analogdevicesinc/hdl #1695: [BUG] AXI SPI Engine returns junk in higher bits when using bits-per-word less than data_width

dlech, 2025-04-30:
**Describe the bug**
I'm using a project (AD7606C-18) that reads 18-bits per word. This means that we get 4 bytes for each 18-bit word read from the SPI controller. The Linux driver for this and many other expect the bits beyond the 18 to be all zeros. However, the appear to contain random data.

**To Reproduce**
Steps to reproduce the behavior:
1. Build HDL from the commit below.
2. Use mainline kernel ad7606 driver and read the `in_voltage0_raw` attribute. This shows a value in the like 9000 volts!

**Expected behavior**
It should show something between 0 and 2.5V.

**Screenshots**
Here is the actual data read as seen by a logic analyses.
[image]

Compared to the data read in Linux, we can see the low byes (little-endian order) match but most of the upper bytes are junk.

```console
root@zed-dlech:~# iio_readdev ad7606c18 -b 8 -s 8  | hd
00000000  68 9f 00 00 7f 93 a0 7d  21 63 fc 4d 42 63 84 8c  |h......}!c.MBc..|
00000010  14 63 08 8d 25 63 50 8c  f2 62 94 8c 52 68 c8 8b  |.c..%cP..b..Rh..|
...
```

First sample is ` 68 9f 00 00`. This is actually what we are expecting, higher bits are all 0.

Second sample is `7f 93 a0 7d`. This is the problem, `a0 7d` should be `00 00`. Same issue can be seen with all of the later samples.

**Desktop (please complete the following information):**
 - Project name and used carrier board: ad7606x_fmz/zed NUM_OF_SDI=1 ADC_N_BITS=18 INTF=1
 - Used Software: Linux
 - Tool version Vivado 24.2
 - HDL Release version: Commit 160cd8e4a (main from a few days ago, no SPI engine changes since then)
 - Software Release version: N/A

dlech, 2025-05-01:
After staring at this a bit more, it looks like this issue is that the data register is cleared after every CS assert, but not after ever word read. Since we are reading 8 18-bit words without releasing the CS line, only the first word is zeroed but the rest aren't.

https://github.com/analogdevicesinc/hdl/blob/0ddf23c7382af9164ae08eed2c275f6ae8ecd730/library/spi_engine/spi_engine_execution/spi_engine_execution_shiftreg.v#L263-L271

This can be seen in the data capture where every other line has a couple `00 00` at the beginning where the CS assert cleared the register.

```console
00000000  68 9f 00 00 7f 93 a0 7d  21 63 fc 4d 42 63 84 8c  |h......}!c.MBc..|
00000010  14 63 08 8d 25 63 50 8c  f2 62 94 8c 52 68 c8 8b  |.c..%cP..b..Rh..|
00000020  5f 9f 00 00 5c a7 7c 7d  20 63 70 9d 3e 63 80 8c  |_...\.|} cp.>c..|
00000030  14 63 f8 8c 22 63 50 8c  f0 62 88 8c 4d 68 c0 8b  |.c.."cP..b..Mh..|
00000040  5e 9f 00 00 1f bb 78 7d  21 63 7c ec 3f 63 84 8c  |^.....x}!c|.?c..|
00000050  12 63 fc 8c 25 63 48 8c  ef 62 94 8c 4e 68 bc 8b  |.c..%cH..b..Nh..|
```

LBFFilho, 2025-05-08:
As discussed, we'll keep the current behavior (unused bits are undefined), and masking will be done in software

closed 2025-05-08 by LBFFilho